bambulab/BambuStudio #4416: Keeps asking for P1S access code

carlcastro, 2024-07-02:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Printer

P1S

### How to reproduce

Device
Try to connect to printer
Printer requests access code every time

### Actual results

BambuStudio doesn't remember access code and requests it each time

### Expected results

Expect Bambu Studio to remember printer's access code

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-07-03:
HI，[carlcastro](https://github.com/carlcastro)
Could you please provide a video or git，we will check this。
Thanks

carlcastro, 2024-07-03:
Just in Bambu Studio, when clicking printer here. My printer always goes
into this locked mode and requires me to input the access code.

Before it would remember the access code and never need to have it
re-entered.

[image: image.png]

herrdeh, 2024-11-01:
A clean installed v 1.10.0.81 on ubuntu Linux has got the same issue.