bambulab/BambuStudio #4419: Latest AMD drivers (24.6.1) causes Prepare window in Bambu Studio to be black

kenneaal, 2024-07-02:
### Bambu Studio Version

1.9.1.66+

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: Ryzen 7950X3D GPU: AMD RX 7900XT

### Printer

Bambu Lab X1C

### How to reproduce

Install AMD Adrenalin 24.6.1. (Driver Version 24.10.21.01-240620a-404579C-AMD-Software-Adrenalin-Edition)
After installation, Prepare screen in Bambu Studio is black. You can still interact with it, just does not display.

### Actual results

[image]
Prepare window is black

### Expected results

Normal prepare window; the Preview window still works.
[image]

### Project file & Debug log uploads

Not relevant to issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-07-03:
We are currently considering upgrading the shader version, please be patient and wait~

Haidiye00, 2024-07-04:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0 
Can you try this Prusa package and see if the 3D page opened by the software is working properly.thanks

kenneaal, 2024-07-05:
No, PrusaSlicer also presents a black screen with the latest AMD drivers. 
[image]
Log file included. It has also been tested with the latest Bambu Studio release; I rolled it back for debugging.

[debug_Fri_Jul_05_17_37_54_37876.log](https://github.com/user-attachments/files/16111619/debug_Fri_Jul_05_17_37_54_37876.log)

CrAaAaZyLeGs, 2024-07-08:
same problem! any solution yet?

soluna2020, 2024-07-16:
I have the same setup. Open AMD Adrenalin. Go to Gaming main tab. Then to Graphics sub tab. Turn on HYPR-RX. The preview screen will then be visible. Enjoy!

Haidiye00, 2024-07-17:
Can you try upgrading AMD's drivers?

OmiCron07, 2024-07-24:
Has anyone tested with the new version 24.7.1 ?

kenneaal, 2024-07-30:
AMD Adrenalin 24.7.1 resolves this problem, however the problem is still likely a factor that may reoccur at a later date if not resolved in Bambu Studio/PrusaSlicer now.

bepstein111, 2024-08-02:
Same issue with NVIDIA drivers in Linux with Compiz but NOT with kwin.

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95