bambulab/BambuStudio #4420: Brim (Brim-Object gap?) doesn't account for Elephants-foot compensation...this creates a gap between brim and object

PhilBaz, 2024-07-02:
### Bambu Studio Version

current 

### Where is the application from?

Bambu Lab Official website

### OS version

win 10

### Printer

X1

### How to reproduce

https://github.com/SoftFever/OrcaSlicer/issues/3720   see thread via Orca. Im sorry, the current approach is just wrong. It should 'Just work'.

Add elephants-foot compensation and brim and slice.... If Brim-Object gap is zero there should be no gap. but there is! This gap is caused be Elephant foot compensation, and can be closed by reducing this to zero.

### Actual results

There is a gap

### Expected results

No gap, brim should touch, or be controlled entirely by brim-object gap

### Project file & Debug log uploads

[if-nfs-a-healthy-lifestyle-to-improve-cholesterol-triglycerides.zip](https://github.com/user-attachments/files/16071794/if-nfs-a-healthy-lifestyle-to-improve-cholesterol-triglycerides.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

hotellonely, 2025-04-10:
Is this going to be fixed anytime in the future? It makes the brim function totally unreliable and meaningless. @QingZhangBambu

PhilBaz, 2025-04-10:
@hotellonely Completely agreed. Cura has had this right for like a decade.

frank-pian, 2025-09-05:
I looked into the issue: elephant’s foot already affects the outline when the first layer is drawn, while the brim is generated later through outline offsetting. This causes gaps that are difficult to fix.

That’s also why Prusa and Orca face the same problem, whereas Cura seems to use a completely different slicing logic.

https://github.com/bambulab/BambuStudio/blob/7e075f71be1516c110f9952d6a0b103d10a4777d/src/libslic3r/PrintObjectSlice.cpp#L1206-L1242