bambulab/BambuStudio #4423: 布尔差集功能在处理有较多孔洞的模型部位时不生效

moe001, 2024-07-03:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

10940x/32g/3090

### Printer

P1SC

### How to reproduce

1. 载入附件包含的3mf
2. 在对象界面选中唯一的组合体
3. 点击预览界面上方的**布尔运算-差集**
4. 从中减去选择**绿色正方体**，与之相减选择剩下的那个**曲线模型**
5. 勾选**删除输入**，点击**差集**按钮以开始处理
6. 处理完成后可见**绿色正方体**毫发无损，差集没有生效（见下述Actual results）
7. 将**绿色正方体**向打印盘边缘平移，直到正方体至少有一个立面已完全不与曲线模型相交
8. 再次重复2~5步，可见**绿色正方体**上正确的留下了曲线痕迹，差集正确生效（见下述Expected results）

### Actual results

**绿色正方体**完好无损，使用切片功能也可确认其中不包含空心部分
[image]

### Expected results

正确执行差集操作（图为曲线较为简单的部分，从而能得到正确结果，只作为示例，曲线复杂的部分因当前bug无法得到结果）
[image]

### Project file & Debug log uploads

[240703bug.zip](https://github.com/user-attachments/files/16075862/240703bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-07-04:
Studio右下角给出了布尔运算失败的原因。建议将circle_pattern.stl转为负零件，再直接切片。
[image]

moe001, 2024-07-04:
[image]
如图所示我这边无法看到这个错误提示，这是否是另一个bug亦或是我的某些设置影响？如果是后者的话我如何启用这类提示以发现一些已知的错误？