bambulab/BambuStudio #4427: Filament wont sync

friikadello, 2024-07-04:
### Bambu Studio Version

1.9.35.50

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Additional system information

amd 5800x3d 4070 ti super 32gb ddr4 ram

### Printer

a1 and p1s 

### How to reproduce

I chose the correct printer A1 in this case but it shows wrong filament colors. I click resync, but it still shows p1s colors. I close the bambu studio and reapon, but still wont load correct colors

### Actual results

[image]
I have purple, white, blue and black in A1 Green, Black, Orange and REd is in p1s

### Expected results

After resync i would expect to get correct colors in a1

### Project file & Debug log uploads

[debug_network_Thu_Jul_04_03_37_08.log.zip](https://github.com/user-attachments/files/16092368/debug_network_Thu_Jul_04_03_37_08.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

friikadello, 2024-07-04:
I cant fix it in Prepare Tab, had to to go to device chose a1 there and only then in prepare i could resync. very confusing for noobs like me.

ttt888zzz, 2024-07-04:
Please have a try：
[image]