bambulab/BambuStudio #4430: TPU filaments do not show up in AMS settings

CNCdood, 2024-07-04:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Pro

### Additional system information

AMD Ryzen 5 1500X Quad-Core Processor             3.50 GHz
16.0 GB RAM

### Printer

P1S

### How to reproduce

Go to device. Click on external spool and try to install "Generic TPU" as the outer spool.

### Actual results

closed and reopened app and results are the same. Even though the filament is checked in the filament settings of the "prepare" screen, its not able to be defined in the "Device" screen.

### Expected results

To be able to tell my machine that there's TPU in the external spool.

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CNCdood, 2024-07-04:
[image]

ttt888zzz, 2024-07-05:
Please try the following steps：
[image]
[image]

CNCdood, 2024-07-05:
> Please try the following steps：
> 2024-07-05_121235.png (view on web)
> 2024-07-05_121503.png (view on web)

it didnt work