bambulab/BambuStudio #4433: 在复杂曲线模型上应用特定数值的“X-Y 孔洞尺寸补偿”会导致程序崩溃

moe001, 2024-07-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

10940x/32g/3090

### Printer

P1SC

### How to reproduce

1. 载入附带的工程文件
2. 不修改任何设置，直接切片

### Actual results

软件会在数秒后直接无响应或消失。

偶发的，不会无响应或消失，但会弹出如下提示框，点击确认之后，软件会处于不正常的状态，无法再次进行切片等操作。
[image]

### Expected results

正常产生切片结果，或提示无法切片的原因，而不是直接崩溃。

补充：
在“X-Y 孔洞尺寸补偿”填写其他值时，无论是比`-1`更大或者更小的值，如`-0.4`或`-2`，都能够正确切片。

### Project file & Debug log uploads

[240705bug.zip](https://github.com/user-attachments/files/16105108/240705bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-08-14:
听起来是缩放轮廓的函数操作有点问题，我会检查这部分。不过涉及到这部分不是很好修复。

QingZhangBambu, 2024-10-12:
暂时没在代码上看出什么问题，也没有复现。下周或者下一周会发布新版，您可以试试是否更稳定。后面如果有必现的文件可以继续上传