bambulab/BambuStudio #4434: 对于特定的 stp 文件，Bambu Studio 导入时出错

deepbluez, 2024-07-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU：R9 7950X 3D，NVidia RTX 4090，Memory 128 GB

### Printer

X1E

### How to reproduce

1. 用 Fusion 360 分别导出 step 和 3mf 格式
2. 导入支架简易版底座 v1.step，显示文件破损，非流形边
3. 导入支架简易版底座 v1.3mf，文件正常可打印
[支架简易版底座 v1.zip](https://github.com/user-attachments/files/16108300/v1.zip)

### Actual results

特定格式的 step 文件打开后显示异常

### Expected results

step 文件可以正常打开

### Project file & Debug log uploads

[底盘支架 v1.zip](https://github.com/user-attachments/files/16108329/v1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MackBambu, 2024-07-10:
这个step文件拓扑表达显然是错误的，可能是网格模型反向转化的step模型，会有较多错误，建议直接使用mesh模型打印。
[image]

deepbluez, 2024-07-10:
[支架简易版底座 v1.zip](https://github.com/user-attachments/files/16155249/v1.zip)
这个文件是 Fusion 360 创建的，附上原始文件。

MackBambu, 2024-07-10:
减小倒角半径避免与边界相交可以消除拓扑错误，可以像fushion360报告这个问题
[image]

deepbluez, 2024-07-10:
明白了，还好这个问题有个曲线救国的方法，用 Fusion 导出 3mf 格式，打印是一切正常的

closed 2024-07-10 by deepbluez