bambulab/BambuStudio #4436: [Mac OS] Can't drag & drop object to change order

TheLazyHatGuy, 2024-07-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Additional system information

Mac Mini 2023

### Printer

P1A

### How to reproduce

1. Add multiple objects
2. Change to Objects process
3. Change Print Sequence to By Object
4. Try to drag and drop one object above the other and change sequence

### Actual results

Instead of being able to change the order of objects, objects are just highlighted

### Expected results

Can use drag and drop to change sequence order

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/16110256/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TheLazyHatGuy, 2024-07-05:
Similar bug is also present on the Linux version 1.9.3.50

LiZ-Li-BBL, 2024-07-15:
Thanks for reporting this, We'll try to look at it.