bambulab/BambuStudio #4437: Load filament user presets to AMS

weenan, 2024-07-05:
When setting up a preset for a non-BL filament it is only possible to choose that preset in the project and not for the AMS.

I would like it if the user presets were available to choose on the Device tab when choosing filaments for the AMS.

johnny-bit, 2024-07-05:
You can.
When selecting filament to use, create new filament, then set up all it's params and copy from your preset to it. That way it'll be visible in device.

MadMax389, 2024-07-06:
Create a custom filament. Those will be available on the device screen.
https://wiki.bambulab.com/en/bambu-studio/create-filament
[image]