bambulab/BambuStudio #4438: Studio Quits Unexpectedly

sisuwest, 2024-07-05:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

A1

### How to reproduce

Often after sending a file to a printer (it doesn't matter which model) once the countdown is finished the app quits unexpectedly. 

### Actual results

Often after sending a file to a printer (it doesn't matter which model) once the countdown is finished the app quits unexpectedly. 

### Expected results

Don't quit

### Project file & Debug log uploads

No log file exported 

### Checklist of files to include

- [X] Log file
- [ ] Project file

freakintoddles2, 2024-07-12:
I have the same issue - it seems random. I would say, 80% of the time it's fine, 20% it quits unexpectedly immediately after sending the file to the printer. It's always right after the countdown to switch to the Device view that happens automatically.

walterwongbbl, 2024-07-26:
Hi.  @sisuwest 
Can you provide more information, such as Studio logs and crash dmp files