bambulab/BambuStudio #4441: Select 2 colours for filament

poudenes, 2024-07-06:
Hi All,

Would be nice to have the option to select 2 colours for multicolour filament.  Or even better 3. Then icons look like this:

[image]

Now you have to select a colour and in the Studio when you sync the AMS you not sure what filament is loaded.

Zeustopher, 2024-08-10:
I desperately want this. I bought quite a few dual-color and tri-color silk filaments and keeping them straight in the AMS / slicer is difficult. Having this ability to visually see that it is one of these will help make sure I'm selecting the correct one.

poudenes, 2024-12-16:
Doesn't seem a must have for BambuLabs to implement this...

poudenes, 2025-07-23:
no priority for Bambu. Will close this

closed 2025-07-23 by poudenes