bambulab/BambuStudio #4442: I cannot start the prints with abs, keep getting errors

razvanrsk, 2024-07-06:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor: AMD Ryzen 7 5800X 8-Core Processor              (16 CPUs), ~3.8GHz
                   Memory: 32768MB RAM
      Available OS Memory: 32668MB RAM
 Card name: NVIDIA GeForce RTX 4070

### Printer

P1S

### How to reproduce

I tried to calibrate temperature for abs, looked on maker lab for a temp tower and how to do it, i printed 2 times, failed. i look for other temp towers, while looking for others, when i tried to start the print with slot 1(from ams) it says "NOTE: Only the ams slots loaded with the same material type can be selected"
i tried to calibrate and recalibrate, nothing changes
PS: when i was writing this and trying to recreate, i opened a new bambulab studio and tried a stl from me(not from makerworld) and the only thig showed in the ams selection slot is ONLY ABS, (slot 1) and i have 2-3-4 full with pla

### Actual results

[image]
[image]
[image]

### Expected results

Let me select all the slots i want to print with

### Project file & Debug log uploads

[Iteratie (3).zip](https://github.com/user-attachments/files/16115133/Iteratie.3.zip)
[Temp_Tower_ABS_ASA (1).zip](https://github.com/user-attachments/files/16115134/Temp_Tower_ABS_ASA.1.zip)
[debug_Fri_Jul_05_23_25_29_23636.log.zip](https://github.com/user-attachments/files/16115138/debug_Fri_Jul_05_23_25_29_23636.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

razvanrsk, 2024-07-06:
i think i found the problem.
1. Go to preview, select the first filament type( if abs filament is select as the first slot, doesnt matter if the 2nd ,3th, 4th is pla, it will not show up and give an error. if i change the first filament to pla,, the other slots will be available to be selected.)

walterwongbbl, 2024-07-15:
@razvanrsk 
Hi. Do you intend to use PLA to print the ABS preset that you have set up?