bambulab/BambuStudio #4443: 拓竹切片软件在切片大一点或者复杂一点的模型时会闪退或者蓝屏死机

osmant123, 2024-07-06:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

版本	Windows 11 专业版 版本号	23H2 安装日期	‎2024/‎5/‎19 操作系统版本	22631.3737 体验	Windows Feature Experience Pack 1000.22700.1009.0

### Additional system information

	
电脑型号	华硕 台式电脑  (扫描时间：2024年07月06日)
操作系统	Windows 10 专业版 64位 ( DirectX 12 )	
处理器	12th Gen Intel Core i9-12900K
主板	华硕 ROG MAXIMUS Z690 HERO ( 英特尔 未知(4660h) )
内存	64 GB ( GLOWAY )
主硬盘	西数 WD_BLACK SN850X 1000GB ( 1 TB / 固态硬盘 )
主显卡	Nvidia GeForce RTX 3080 Ti ( 12 GB / 技嘉 )
显示器	戴尔 DELA0A6 DELL U3415W ( 34.1 英寸  )
声卡	英特尔 未知(7AD0h)
网卡	英特尔 Wi-Fi 6E AX210 160MHz

### Printer

Bambu Lab P1S

### How to reproduce

点击切片单盘后进度卡在15%，然后要么是闪退要么是蓝屏死机。

### Actual results

软件闪退或者电脑蓝屏死机

### Expected results

应该完成切片步骤

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16115387/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ttt888zzz, 2024-07-08:
可以请您帮忙上传一下源文件吗，帮助排查下这个问题

QingZhangBambu, 2024-07-08:
@osmant123 您好，已知有时候CPU对多线程的支持不够稳定，容易产生崩溃问题。请您尝试调整一下连接下的方法来帮助软件正常运行  Bambu Studio crashes when trying to slice project · Issue #3318 · bambulab/BambuStudio