bambulab/BambuStudio #4446: STOP USING 90C For ABS BED TEMP.  IT DESTROYS EVERY PRINT

RiskyJoint, 2024-07-06:
### Bambu Studio Version

1.8.4.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

LIFE IN THE SOUTH, USE ABS FILAMENT AND CLICK PRINT.  EVERY SINGLE PRINT THAT HAS FAILED ON THIS PRINTER SOLEY IS BECAUSE YOUR SLICER USES 90C FOR ABS BED TEMP. IT DOES NOT WORK IN FLORIDA, IT DOES NOT WORK FOR ABS HERE. PLEASE STOP USING IT AS DEFAULT OR ALLOW US SO SET THE SLICERS DEFAULT, YOU HAVE COST ME 100.00S BECAUSE OF THIS MISTAKE. 

### Actual results

SETTING TO 110C PERFECTION EVERY TIME

### Expected results

YOU TO STOP THINKING YOUR PRESET VALUES ARE WHAT WORK GLOBALLY AND NOT HAVE TO USE CUSTOM PRESETS ALLOW THE USER TO SET HIS OWN. 

### Project file & Debug log uploads

fix the issue. 

### Checklist of files to include

- [X] Log file
- [X] Project file

XunZhangBambu, 2024-07-12:
Hello, I have forwarded your comments to the colleagues related to filaments.

XunZhangBambu, 2024-07-12:
You can try applying glue.Setting the temperature to 110 degrees will cause many problems.