bambulab/BambuStudio #4450: unprintable stl and step file (orcaslicer does print it)

gitterman, 2024-07-07:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Printer

Bambulab P1S with AMS

### How to reproduce

Just try to load the attached model files.
STEP doesn't load at all, STL has a strange cut in between.
I loaded the STEP to orcaslicer and it did print well.
There is also no problem if I omit the chamfering on the model.

### Actual results

step file gives: The file doesn't contain any geometry data
stl file gives a warning after slicing that there is an empty layer in between the part.
without the chamfer the stl file slices well.

### Expected results

Both step and stl file should load and slice properly

### Project file & Debug log uploads

[Mounts.zip](https://github.com/user-attachments/files/16119312/Mounts.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-07-11:
saugrohrHalter.stl has many errors in the file that can cause it to display incorrectly.

[image]

saugrohrHalter.step and saugrohrHalter_noFillet.stl are loaded correctly. Can you provide error logs for each?

[image]

gitterman, 2024-07-11:
Strangely, I cannot reproduce the error anymore.
Today I could load the step files without problems and even the stl file worked after windows had done some repair on it.
I don't know what has happened.
There was an update for my windows from Microsoft thoug 2 days ago, which may have changed something, though.

closed 2024-07-11 by gitterman