bambulab/BambuStudio #4454: Bambu studio https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide for ubuntu inoperable

dsimonjones, 2024-07-07:
### Bambu Studio Version

trying to install latest on ubuntu

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04.4 LTS

### Printer

none yet

### How to reproduce

Hi:)
I followed the install guide for ubuntu with no luck. parts 2 building dep and 3 building
I am getting cmakelists.txt errors
cmake ../ -DDESTDIR="/home/_username_/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1
CMake Error: The source directory "/home/qwerty/work/projects/BambuStudio/deps" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

### Actual results

https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

### Expected results

https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

### Project file & Debug log uploads

https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-07-08:
@dsimonjones 

could you show your directory list in /home/qwerty/work/projects/BambuStudio?

dsimonjones, 2024-07-11:
Hi Thanks for getting back to me

[image: image.png]
Kindest Regards

Simon

lanewei120, 2024-07-12:
> Hi Thanks for getting back to me [image: image.png] Kindest Regards Simon
> […](#)

@dsimonjones 
it seems your image attached can not be shown correctly

dsimonjones, 2024-07-12:
ok sorry replied Via email. copy paste below
qwerty@qwerty-HP-Pavilion-x360-Convertible:~/work/projects/BambuStudio$ ls -a
.   Bambu_Studio_linux_ubuntu-v01.09.03.50           build  install_dir
..  Bambu_Studio_linux_ubuntu-v01.09.03.50.AppImage  deps

Thanks again for your help:)

lanewei120, 2024-07-13:
can not find source codes in ~/work/projects/BambuStudio

and from your log, there is no CMakeLists.txt in /home/qwerty/work/projects/BambuStudio/deps

are you executing the command 'cmake ../ -DDESTDIR="/home/username/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1' in /home/qwerty/work/projects/BambuStudio/deps/build
and could you check the source codes in /home/qwerty/work/projects/BambuStudio/deps?

just give a snapshot of my directories for your compile
[image]