bambulab/BambuStudio #4455: Model Slicing Crashes out Bambu Studio

David3333333333333333, 2024-07-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

CPU: i9-14900k 3.20GHZ
RAM: 64 GB
GPU: 4080 Super

### Printer

Bambu Lab X1 Carbon Combo

### How to reproduce

You load in the models on to 3 plates and slice them all at once or you can try slicing them one at a time.

### Actual results

Once you click the slicing feature Bambu Studio instantly closes out but still remembers the last changes I did to it.

### Expected results

bambu studio to crash out.

### Project file & Debug log uploads

[GothicTowerMossPole-TheArgentRose-ModularD-Looser.zip](https://github.com/user-attachments/files/16122141/GothicTowerMossPole-TheArgentRose-ModularD-Looser.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ARK3r, 2024-08-20:
Any updates on this?

QingZhangBambu, 2024-08-21:
there are some methods may help, could you try it? the cpu may not stable enough to support multi-thread. 
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

ARK3r, 2024-08-21:
The crashes have been intermittent so far, once I get to a mesh that I need to print and the program crashes, I'll test this out and report back.

ARK3r, 2024-08-21:
I have exactly the same CPU, GPU, and 64GB RAM, and setting the affinity to use only the first 4 cores (CPU0, CPU1, CPU2, CPU3) using [Process Lasso](https://bitsum.com/) as suggested [here](https://www.reddit.com/r/BambuLab/comments/1chbuiy/bambu_studio_crashingfreezing_on_slicing/) has been running without issues, though a little slow during slicing (as expected).

closed 2024-10-12 by QingZhangBambu