bambulab/BambuStudio #4458: Wrong filament type loaded

geo-luki, 2024-07-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

P1S

### How to reproduce

I am preparing the files for printing, I have a PA6-GF filament location in AMS and it is also set for printing. But AMS does load PA6-GF correctly, but then in the view in AMS settings there is only **PA-GF**. So when I send the prepared file via LAN to the SD card in the printer and want to start printing via BambuHandy, I can't because I get an error about an incompatible filament.

### Actual results

[image]
[image]
[image]
[image]
[image]

### Expected results

Of course I restarted the computer, the printer and even tried to put the spool in another slot. But nothing that would help.

### Project file & Debug log uploads

[Bez názvu.zip](https://github.com/user-attachments/files/16133388/Bez.nazvu.zip)
[log.zip](https://github.com/user-attachments/files/16133390/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-07-16:
see：https://github.com/bambulab/Bambu-Handy/issues/205

closed 2024-07-16 by ttt888zzz

LibaoBambu, 2024-07-16:
@geo-luki Please reinstall the App. If the issue persists, let me know.