bambulab/BambuStudio #4459: Adding "change filament" to layer in Preview Slider does not let you manually change filament.

mturley777, 2024-07-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P!P

### How to reproduce

1) Slice a model
2) In Preview, use slider on right to select a layer.
3) Right click mouse over the slider + and select change filament.
4) Slice and print

### Actual results

Model is printed with currently loaded filament and no attempt is made to allow user to change filament.

### Expected results

Model should print with currently loaded filament up to layer containing the layer change.  Then currently loaded filament should be unloaded and user prompted to load next filament.  Also, the prime tower should be used for this model print if selected in settings.

### Project file & Debug log uploads

[No Filament Chg.zip](https://github.com/user-attachments/files/16134650/No.Filament.Chg.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-07-09:
@mturley777  Could you describe your problem in more detail and add some screenshots if possible?

zhimin-zeng-bambulab, 2024-08-06:
@mturley777  Only single-color models are allowed to insert "change filament" in the preview page after slicing.

mturley777, 2024-08-09:
Sure, on a printer that has no AMS connected,  take any single color project that has been setup with multiple filaments:

[image]

Now slice the project and then add a "change filament' on the layer of your choice:

[image]

Now re slice and see that indeed the change filament has been set for the print:

[image]
[image]

Now print, and observe that the "change filament" does nothing.  The entire project is printed in the filament that is already loaded in the printer.

zhimin-zeng-bambulab, 2024-08-12:
@mturley777  Do you want to be able to change colors manually when printing with only external filament?

mturley777, 2024-08-12:
Yes.  Since the function is available, and you can use it to set a filament change, and it shows in the preview that it is set, then it should work to unload the current filament and prompt the user to load the new filament manually.

EricDimitri, 2024-11-15:
@zhimin-zeng-bambulab zhimin Adding to this, I was trying to print with TPU, multivolor, so I was using external filament, and did a change color, nothing happened. I had to use PAUSE instead of CHANGE COLOR to change color. The bad thing is that it doesnt generate the PRIME TOWER, and TPU  needs that otherwise it will lick for sure. Please add this as a functionality, and also add Prime tower at least as an option, so we can clean the nozzle and generate a continous flow of fillament right before continueing printing. THanks!!!

BillHerb, 2025-02-09:
same issue here but as you can see in this screenshot at least his preview tab shows the filament changes color mine doesn't show the black even though at layer 2, I inserted a color change to black it all still shows yellow it's the same bug.  this ticket is over a year old is this going to get fixed sometime soon? Also while your at it how come every time you print something you have to hit the re-sync filament button to grab what colors are in the ams it should remember these.  

[image]

[image]

BillHerb, 2025-02-09:
also to be clear i'm using ams and it still doesn't work a manual review of the exported gcode also reveals there is no code entered to actually tell the printer to change the filament