bambulab/BambuStudio #4460: Painted model does not print in multicolor (no filament change(s) made)

mturley777, 2024-07-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1)  Click on model and select Paint Tool,
2)  Paint parts of the model in a different filament.
3) Slice and print model.

### Actual results

Model is completely printed with the currently loaded filament.

### Expected results

Model should print with currently loaded filament up to layer(s) containing the painted surfaces on the model. Then the user should be prompted to change filament as needed to actually "Paint" the object with multiple filaments.  Also, the prime tower should be used for this model print if selected in settings.

### Project file & Debug log uploads

[Painted but no filament change.zip](https://github.com/user-attachments/files/16134689/Painted.but.no.filament.change.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-07-09:
Please try following the steps below
[image]

mturley777, 2024-07-10:
I happen to be printing this with a 98A TPU which I have printed a bunch in the past from the AMS.  Now however it will jam in the AMS every time it unloads the filament (the 98A TPU).  Not sure why the AMS will no longer handle the TPU, but to get this job printed I want to manually change the filament when needed.

So when trying this with the AMS disconnected, or unselecting the "Use AMS" checkbox when submitting the print, the model is always printed in just the filament that is manually loaded from the side spool.

So regardless of whether I use change filament at layer, or painting the model with a different color filament, the model is shown on screen colored correctly, but it will never prompt for any filament changes and prints in a single color.  

If this functionality is not supported without the AMS, then the change filament selection and painting operations should not be available to the user when they do not have the AMS, or do not have it connected.  It would be great however if the machine gcode for "Change Filament", and painting functionality, was coded correctly to handle both manual filament loading and AMS loading.

Haidiye00, 2024-09-19:
Hello, can you record your operation GIF? Thank you