bambulab/BambuStudio #4461: 'Enable AMS' checkbox shown as 'Enable' only under Linux

papersail, 2024-07-08:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Pop_OS 22.04 LTS

### Additional system information

Dynabook PORTEGE x30l-j laptop

### Printer

Bambu Lab P1S

### How to reproduce

1. Under Preview, click 'Slice Plate'
2. Click 'Print Plate', which will show the dialog

### Actual results

The 'Enable AMS' checkbox only display 'Enable', where 'AMS' is missing
[image]

### Expected results

'Enable AMS' displays as it should

### Project file & Debug log uploads

[debug_bambu_studio.zip](https://github.com/user-attachments/files/16134939/debug_bambu_studio.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

LiZ-Li-BBL, 2024-07-09:
Thank you for reporting this issue, we will fix it.

nivek1385, 2025-02-24:
I'm still experiencing this on LinuxMint using Bambu Studio Ubuntu 24.04_v01.10.01.50.AppImage