bambulab/BambuStudio #4468: printer a1 mini is over purging filament since last firmware

lebelge-a1mini, 2024-07-10:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Ventura 13.6.7

### Printer

A1 Mini

### How to reproduce

on all points

### Actual results

Work ok but purges filament at least 3 to time morethan before...it's extremely wastful and risk cloging

### Expected results

fix the issue and go back to previous purging amount

### Project file & Debug log uploads

G1 E-0.8 F1800 ; retract is the current value ( I think )
[0309CA440300553_07-10_11_32_22.705_v01.03.00.00_idx_6.log](https://github.com/user-attachments/files/16163781/0309CA440300553_07-10_11_32_22.705_v01.03.00.00_idx_6.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lebelge-a1mini, 2024-07-10:
How to reproduce:  on all job sent to the printer it purges 3 to 4 times more than before

lebelge-a1mini, 2024-07-10:
By the way it purges 2 times,  the second one is large and totally unessessary

XunZhangBambu, 2024-07-15:
@lebelge-a1mini Do you mean the garbage that's been flushed out all  or that the number of flushes has increased by 3 to 4 times?

lebelge-a1mini, 2024-07-15:
I mean that when it starts its process it poops , draw the line of filament and then it does it again..but the second time it really bad.   the poop is uge, it is 3to times bigger thm normal.  I have a P1S and a A1 and they both work the same... ghe mini started just like the other ones...but ater the last update.  the volume of poop in thr second phase is way,,,way to much