bambulab/BambuStudio #4470: [Feature Request] Adding fan speed control for ironing layers.

BambuLabDiscord, 2024-07-10:
Cooling has a massive impact on the quality of an ironing layer on some filaments. I've found with Jayo black matte filament turning of cooling completely during ironing gives great results, while having cooling on gives worse results than without ironing at all as seen in this picture:
https://imgur.com/a/C6WSQub

Left side is not ironed, middle is ironed with 0% fan speed and right side is ironed with 60% fan speed.

caizixian, 2024-10-19:
Second this. See also https://github.com/SoftFever/OrcaSlicer/issues/6979

caizixian, 2024-10-19:
Having the same problem with eSUN PLA+. Ironing with 0% fan (middle section of the photo) gives the best result. 
[image]