bambulab/BambuStudio #4471: [Color Picker, Color Hex Codes, and Color Database] Color Picker, Color Hex Codes, and Color Database

BambuLabDiscord, 2024-07-10:
One of your biggest selling points is multicolor; yet, I have to keep entering RGB values... Please add a Color Picker, Color Hex Codes, and a Color Database for saving different filament color properties! 

A great example of a Color Picker and Color Hex Codes can be found in Fusion 360. I also Tweeted an image: fatboy1271

fatboy1271, 2024-07-10:
[image]

indyjason79, 2024-11-19:
Yes, we need this very much please! I'm getting tired of having to convert RGB/hex.

InterruptionF10, 2024-12-05:
Agreed! Just bought 8 new Bambu Labs colors from their recent sale. Now having to convert their Hex code values to RBG for a new project I am working on. I want my assembly in the .3mf to closely match the colors I am going to see when I print and post it.