bambulab/BambuStudio #4472: STL no longer imported as assembly

cutmoney, 2024-07-10:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.5

### Additional system information

Macbook Pro M2 Max

### Printer

P1P and A1

### How to reproduce

1. Import
2. Choose STL
3. Used to bring up a message asking if you wanted to import as a group/assembly, now that no longer exists. You have to manually assemble after import.

### Actual results

Imports objects as multiple bodies.

### Expected results

Option to import as assembly as was previously the case.

### Project file & Debug log uploads

Applies to all projects and STL files.

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye01, 2024-07-12:
Can you provide the STL file? Thank you

Haidiye00, 2024-07-15:
[image]
I tested it and there were no issues

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00

Haidiye01, 2024-10-16:
ok,i have received your letter ,thank you!