bambulab/BambuStudio #4481: Live555 Server TLS Handshake Failure in Bambu Lab X1C Printer Firmware

Fusl, 2024-07-12:
### Bambu Studio Version

-

### Where is the application from?

Bambu Lab Official website

### OS version

-

### Additional system information

I'm experiencing a persistent issue with the Live555 server bundled in the Bambu Lab X1C printer firmware. The issue arises after an extended uptime of the RTSPS stream. Specifically, when the connection is closed from the RTSPS client, the Live555 server hangs and becomes unresponsive to further requests. Subsequent connection attempts fail to complete the TLS handshake for the RTSPS connection, resulting in streaming failures. This issue is independent of any third-party software and is a defect within the printer firmware itself.

**Steps to Reproduce:**
1. Start the RTSPS stream from the Bambu Lab X1C printer.
2. Allow the stream to run for an extended period.
3. Close the RTSPS client connection.
4. Attempt to reconnect using `ffmpeg` or `openssl s_client`.

**Expected Behavior:**
The TLS handshake should complete successfully, allowing the RTSPS stream to be re-established.

**Actual Behavior:**
The TLS handshake fails, and the Live555 server hangs, becoming unresponsive to any further connection attempts.

**Logs & Additional Information:**
- Command used for testing: `ffmpeg -loglevel trace -i rtsps://bblp:password@198.19.81.54/streaming/live/1 -c copy -map 0 -f null -`
- Command used for testing: `openssl s_client -connect 198.19.81.54:322`
- The issue does not occur with the Bambu Studio app, which uses the `bambu_networking.dll` library and bypasses the RTSPS connection.
- This issue affects integration with other systems and software that rely on the RTSPS stream.

**Relevant Links:**
- [Similar Issue Reported](https://github.com/AlexxIT/go2rtc/issues/674)

**Notes:**
- The printer firmware is closed source, and root access is not allowed, limiting further debugging capabilities.
- The issue is not related to Bambu Studio or third-party software but is inherent to the printer's firmware.

**Support Ticket Reference:**
I am aware that this repository might not be the appropriate place to report this issue, but I was instructed to do so in support ticket US240707678002, where the support team refused to assist further.

Please escalate this to your development team to investigate and address the stability of the Live555 server and RTSPS stream.

Thank you for your attention to this matter.

### Printer

Bambu Lab X1C

### How to reproduce

-

### Actual results

-

### Expected results

-

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

beareogaming, 2024-07-13:
I also have this issue. After an extended period of time the stream will cut out and on occasion it will also randomly turn off the lan mode live view check box in the printer settings on it's own

gaolegao-lx, 2024-07-16:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Fusl, 2024-07-16:
I have already provided all the information including printer logs in ticket US240707678002 that I mentioned in my bug report text.