bambulab/BambuStudio #4486: Render error

OmegaII, 2024-07-13:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

AMD Ryzen 9 5900x
32 GB RAM
RTX 3080

### Printer

Bambu X1C

### How to reproduce

Download the model https://makerworld.com/en/models/446204?from=search#profileId-352761
Put it in Bambu Studio
Set the nozzle to 0.2mm
Slice the part

### Actual results

Half of the object is rendered as full. The mandale image just stops mid way, while it does print the full pattern 

### Expected results

That the pattern can be seen on the whole face.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16202662/log.zip)
[mandala-coaster-01_bambu_export.zip](https://github.com/user-attachments/files/16202663/mandala-coaster-01_bambu_export.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-07-16:
[image]
Hello, this 3mf covers a circular disc on the pattern, so it cannot be seen. no problem

zhimin-zeng-bambulab, 2024-08-06:
[image]

The model has non-popular edges, which may lead to incorrect final slicing results.