bambulab/BambuStudio #4487: tree supports brim width

Duudeeee, 2024-07-13:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

i7
16g
gtx1080 8g

### Printer

X1C

### How to reproduce

just change brim support from0 to x 

### Actual results

dont work 

### Expected results

should increase the brim and wont

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-08-12:
We have fixed this issue internally. Please wait for the next release.

Duudeeee, 2024-08-26:
> We have fixed this issue internally. Please wait for the next release.

its not fixed. the option has vanish

ttt888zzz, 2024-08-27:
Please try updating the latest version 1.9.4
https://bambulab.com/en/download/studio

Duudeeee, 2024-08-27:
I

> HI,[Duudeeee](https://github.com/Duudeeee) Please try updating the latest version 1.9.4 https://bambulab.com/en/download/studio

Im on last version. And that options has vanish

MartinChan3, 2024-08-27:
@Duudeeee In V1.9.4, ‘Tree support brim width' has been substituted with ‘Initial layer expansion’,this question was sovled with #4487 under this option.
[image]

Duudeeee, 2024-08-27:
> @Duudeeee In V1.9.4, ‘Tree support brim width' has been substituted with ‘Initial layer expansion’,this question was sovled with #4066 under this option. ![img_v3_02do_c7f091df-6f19-47c8-8d17-65fe2631a4cg](https://private-user-images.githubusercontent.com/19528955/361735425-2afed5de-2124-4a87-b8eb-40fd31fe6630.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ3NTA1NDAsIm5iZiI6MTcyNDc1MDI0MCwicGF0aCI6Ii8xOTUyODk1NS8zNjE3MzU0MjUtMmFmZWQ1ZGUtMjEyNC00YTg3LWI4ZWItNDBmZDMxZmU2NjMwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI3VDA5MTcyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI1NDc3NTkyN2MzNTdmZTU3ZDhmYWRhYTdiMzdhZmI2ZDkxZTA4YjE2MGIxNDk5ZGRhNzdlYzBmZGQ0NGI3OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.VCrRppyZmpixIV8fFK2HPHigxwOOyz4UA9FVTXsGezw)

Nobody Guess u dont say that any place..

closed 2024-08-28 by ttt888zzz