bambulab/BambuStudio #4492: Support "On build plate only" does not work.

NVNDO, 2024-07-14:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Create manual support in the following area: 
[image]

Slice and select "On build plate only"
[image]
 Though, the support is still created on the model.

[image]
[image]

[image]

### Actual results

Even though "On build plate only" is selected in the support section, support is printed on the model

### Expected results

support should not be printed on model when "On build plate only" is selected.

### Project file & Debug log uploads

[Support not working properly.zip](https://github.com/user-attachments/files/16227874/Support.not.working.properly.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-09-18:
It's not a bug. You have painted support enforcers, which has higher priority than "support on build plate only" option in normal support. There are several solutions:
1. remove the painted support enforcers and use auto support;
2. use tree support

NVNDO, 2024-10-12:
> It's not a bug. You have painted support enforcers, which has higher priority than "support on build plate only" option in normal support. There are several solutions:
> 
> 1. remove the painted support enforcers and use auto support;
> 
> 2. use tree support

But I don't get the desired result with either of those options. 

I eventually printed it in Orca Slicer and got the result I wanted.