bambulab/BambuStudio #4493: Issue with Z-hop Types During Retraction to the Purge Tower in 3D Printing

amzaldua, 2024-07-14:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS Sonoma 14.5

### Printer

Bambu Lab A1 series

### How to reproduce

1. Load the 3mf program file attached.
2. Set the Z-hop type to ‘normal’ in printer profile/extruder settings.
3. Start the print and observe the behavior when the printer moves to the purge tower after filament extrusion.
4. Repeat the steps with Z-hop types set to ‘slope’, ‘spiral’, and ‘automatic’.
5. Observe and compare the behaviors in each mode.

### Actual results

In normal mode, after finishing filament extrusion, the printer lifts the Z-axis before moving to the purge tower. In slope, spiral, and automatic modes, no Z-axis retraction occurs; the printer moves directly to the purge tower, grazing the print, and performs the retraction movement upon reaching the purge tower.

### Expected results

In all Z-hop modes, the printer should lift the Z-axis before moving to the purge tower to avoid grazing the print and ensure consistent print quality.

NOTE:
This issue is not observable during the preview of the print paths; it is only noticeable during the actual 3D printing process
The videos are too large. How can I provide them to you?

### Project file & Debug log uploads

[Honda keychain test.3mf.zip](https://github.com/user-attachments/files/16228562/Honda.keychain.test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

te7suo, 2024-07-15:
I have the same problem

Jesusfran94, 2024-07-15:
I have the same problem, finally the topic is talked about. I hope they resolve it soon.

gaolegao-lx, 2024-07-16:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

amzaldua, 2024-07-16:
Hi Chloe,
The ticket number is US240716284001
Regards.

foxjie, 2024-07-16:
yes,My A1 also have the same problem!

amzaldua, 2024-07-23:
Hi @ChloeShang123, any updates here?

tsmith35, 2024-08-27:
I've noticed more and more discussions and complaints from users in the past few months, all related to Z-hop issues. I just read half a dozen discussions (here and elsewhere) related to this bug. Any chance that a solution can be fast-tracked to fix what was broken?

It's taken quite some time just to convince Bambu Lab that there really is an issue, thus there is a large amount of documentation showing that the problem does exist and that it causes both quality and productivity issues.

Gokalex, 2024-10-11:
I have the same issue, every time the head goes to the tower i hear the clicking sounds of the head hitting all the parts in it's path, and just hope that none of them will go flying off (so far every time i print a copule of them detach)
Does setting hop type to normal cause blobs or is it the best fix?