bambulab/BambuStudio #4495: Undocumented printer_error code `0500C011` in Bambu Studio

cjavad, 2024-07-14:
### Bambu Studio Version

01.09.03.00

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Linux

### Printer

Bambu Lab P1P

### How to reproduce

The issue does not happen consistently, and is more so linked to the lack of documentation than an actual error, to reproduce this issue on the P1P printers, simply starting a print from 3rd party clients seems to cause the issue once in a while. The issue is not the starting of the print, but an undocumented error code from Bambu Lab's side, including in Bambu Studio.

### Actual results

When the P1P series printer returns the error code `83935249` or in hex `0500C011` there is no description of the error message.

### Expected results

An error message describing the source of the error should be present, it is not documented even via the error code sources at `https://e.bambulab.com/query.php`

### Project file & Debug log uploads

There are no project specific logs, but the print_error response from the printer follows this format:

```
{"print":{"upload":{"status":"idle","progress":0,"message":""},"nozzle_temper":21.125,"nozzle_target_temper":0,"bed_temper":19.5,"bed_target_temper":0,"chamber_temper":5,"mc_print_stage":"1","heatbreak_fan_speed":"0","cooling_fan_speed":"0","big_fan1_speed":"0","big_fan2_speed":"0","mc_percent":0,"mc_remaining_time":0,"ams_status":0,"ams_rfid_status":6,"hw_switch_state":1,"spd_mag":100,"spd_lvl":2,"print_error":83935249,"lifecycle":"product","wifi_signal":"-52dBm","gcode_state":"IDLE","gcode_file_prepare_percent":"0","queue_number":0,"project_id":"0","profile_id":"0","task_id":"0","subtask_id":"0","subtask_name":"","gcode_file":"","stg":[],"stg_cur":0,"print_type":"idle","home_flag":16904,"mc_print_line_number":"0","mc_print_sub_stage":0,"sdcard":true,"force_upgrade":false,"mess_production_state":"active","layer_num":0,"total_layer_num":0,"hms":[],"online":{"ahb":false,"rfid":false,"version":87449353},"ams":{"ams":[],"ams_exist_bits":"0","tray_exist_bits":"0","tray_is_bbl_bits":"0","tray_tar":"255","tray_now":"254","tray_pre":"254","tray_read_done_bits":"0","tray_reading_bits":"0","version":1,"insert_flag":true,"power_on_flag":false},"ipcam":{"ipcam_dev":"1","ipcam_record":"enable","timelapse":"disable","mode_bits":3},"vt_tray":{"id":"254","tag_uid":"0000000000000000","tray_id_name":"","tray_info_idx":"","tray_type":"","tray_sub_brands":"","tray_color":"00000000","tray_weight":"0","tray_diameter":"0.00","tray_temp":"0","tray_time":"0","bed_temp_type":"0","bed_temp":"0","nozzle_temp_max":"0","nozzle_temp_min":"0","xcam_info":"000000000000000000000000","tray_uuid":"00000000000000000000000000000000","remain":0,"k":0.019999999552965164,"n":1.399999976158142},"lights_report":[{"node":"chamber_light","mode":"off"}],"upgrade_state":{"sequence_id":0,"progress":"","status":"","consistency_request":false,"dis_state":0,"err_code":0,"force_upgrade":false,"message":"","module":"","new_version_state":2,"new_ver_list":[]},"command":"push_status","msg":0,"sequence_id":"19"}} 
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-08-08:
@cjavad 
I'm very sorry, this is actually a non-existent error, we should delete it.

trestrantham, 2024-08-17:
@walterwongbbl Just to confirm here, is this being reported as an error but shouldn't be? I can confirm I am seeing this as well, but on only 1 of my 3 P1S. Is it safe to ignore this as erroneous?

closed 2024-10-17 by walterwongbbl