bambulab/BambuStudio #4496: When printing ABS, the parts at the back of the bed are warping, and not the others

JYPochez, 2024-07-15:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Macos 14.5

### Additional system information

M2 Max

### Printer

X1C and high temp plate

### How to reproduce

trying to print the rosette at : https://www.thingiverse.com/thing:5515942
set to print max items on the bed, so it make 9 prints automatically

### Actual results

the item back left turns spaguetti, and warped
the two other items at the back were warped
all other items are perfect
so may be a problem with high temperatures or fan at the back of the bed ?

### Expected results

9 perfect prints

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/16237975/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-08-01:
OK，we will check this.

QingZhangBambu, 2024-08-01:
@JYPochez Could you re-upload the zip file, it's empty

JYPochez, 2024-08-01:
[2Tdddg1Gv3g.3mf.zip](https://github.com/user-attachments/files/16452562/2Tdddg1Gv3g.3mf.zip)

JYPochez, 2024-08-06:
thanks you for trying to print this. I found it strange that only the models at the back of the printer were bad. but it seems it is not a printer problem. I will try with other tests when I can. thanks again.