bambulab/BambuStudio #4500: Allow to reprint mid print due to a object print failure

Nbr1Sniper, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
If my print fails to stick and I catch it I have to start all over again. 

**Describe the solution you'd like**
Would be nice to reprint an object as long as I'm not too far along in the print layers. 

**Describe alternatives you've considered**
Use a lot of glue before I start a print. Doesn't always work though. 

**Additional context**
3 hours into a print with 99 small objects 5 pieces failed. Caused havoc on the other prints.

SaltWei, 2024-07-30:
@Nbr1Sniper Can not understand

Nbr1Sniper, 2024-07-30:
@SaltWei it would be nice if I could reprint one item or layer if I caught an error mid print without having to throw away and entire print. My example explains my situation.

SaltWei, 2024-07-31:
> it would be nice if I could reprint one item or layer if I caught an error mid print without having to throw away and entire print. My example explains my situation.

Good idea but not easy