bambulab/BambuStudio #4501: problem with skipping with multicolors

sykmer, 2024-07-15:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Place several multicolor items on the printing plate. For example, skip one of the elements halfway and the problem should appear.

### Actual results

When printing several multicolor elements, I had to skip one of them because it broke. After skip, the color sequence was broken and, for example, where the red color was supposed to be, brown was applied, etc. As a result, the entire printing plate went to the trash...

### Expected results

Please investigate such cases and rectify them.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16241149/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-07-22:
@sykmer  Could you share your 3mf file? And which object did you skip?

zhimin-zeng-bambulab, 2024-08-06:
@sykmer  Could you share your 3mf file and which object you skipped?

zhimin-zeng-bambulab, 2024-10-18:
maybe the same as https://github.com/bambulab/BambuStudio/issues/4940

MartinChan3, 2024-10-30:
[1.10 beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81) has solved this problem.

closed 2024-10-30 by zhimin-zeng-bambulab