bambulab/BambuStudio #4502: My toolbar is dancing when my mouse is near them

pebbles822, 2024-07-16:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Device name	DESKTOP-E7HCQAV
Processor	11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz   1.69 GHz
Installed RAM	16.0 GB (15.7 GB usable)
Device ID	E604BA7E-C921-45C0-8D8E-F41B6CF7A99A
Product ID	00325-81953-33878-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points

### Printer

Bambu Lab X1C

### How to reproduce

When I hold my mouse over or near the toolbar all the icons on the tool bar starts jumping. If I stop my mouse pointer over one of the icon it stops moving but as soon as I move the mouse pointer the toolbar starts jumping again. This problem started two updates ago. I attached a short video so you can see what I'm talking about. I don't know where to find a log file for Bambu Lab software so I didn't add it.

https://github.com/user-attachments/assets/2b5947fe-5bb2-4c8b-9a9a-f2ee66fa5741

### Actual results

If I take my mouse pointer away from the Toolbar it stops jumping

### Expected results

I expect my toolbar not to jump like that. its annoying.

### Project file & Debug log uploads

I don't know where to find this information a
[image]
t.

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-07-17:
https://github.com/bambulab/BambuStudio/releases
Can you try a few more historical versions from these links? Is this issue present in all versions, or has it started in which version

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00