bambulab/BambuStudio #4504: A G-code path goes beyond max print height.

Luro02, 2024-07-16:
### Bambu Studio Version

1.9.3

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1S 0.4 nozzle

### How to reproduce

1. Open the attached project file with Bambu Studio
2. Slice the file
3. In the bottom right corner, the error appears:
[image]
(note: it jumps from layer 1230 to some large number that isn't fully visible)

My guess is that this is, because of the variable layer height.

### Actual results

[image]

### Expected results

The file should slice correctly.

### Project file & Debug log uploads

[Pot+boy+planter-upside-down-A1-mini.zip](https://github.com/user-attachments/files/16244767/Pot%2Bboy%2Bplanter-upside-down-A1-mini.zip)
[debug_Tue_Jul_16_09_22_25_17008.log.zip](https://github.com/user-attachments/files/16244799/debug_Tue_Jul_16_09_22_25_17008.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Luro02, 2024-07-20:
[image]
Here is another model, this time without variable layer height enabled. Maybe it is the tree supports?
[shapr3d_export_2024-07-20_09h07m.zip](https://github.com/user-attachments/files/16319186/shapr3d_export_2024-07-20_09h07m.zip)

Cilsor, 2024-08-01:
Hey there. I had a similar issue when trying to print a doll face. Arachne / classic walls, and changing the quality of the adaptive layer height sometimes fixed it, sometimes didn't. **However**,  what fixed it 100% of the time was disabling the new scarf beta feature. My guess is that the scarf feature, which also varies layer height, can have a bug when combined with higher quality adaptive layer height settings. 

Am I right to guess you had scarf joints enabled with these models you had gcode height errors on?

Cilsor, 2024-08-01:
Oh yup, I can definitely see it in image 2 and 3. You were using scarf. I think that's the culprit here. Two different systems trying to come up with variable height layers and one of the two put something out of bounds when working together. 

With scarf being the newer feature, I expect the bug is there.

XunZhangBambu, 2024-08-02:
@Luro02 It seems to be related to arachne wall generator and scarf seam. If you need to continue printing, you can use classical wall generator or disable scardf seam. The specific reason is currently under analysis.

jrimmer, 2024-10-11:
X1c, .24mm Draft, BS 1.9.752 on Mac, cut from larger model that was ~235mm.
[image]
[pendant lamp shade.3mf.zip](https://github.com/user-attachments/files/17346750/pendant.lamp.shade.3mf.zip)