bambulab/BambuStudio #4510: Error Installing BambuStudio on Ubuntu 24.04

CStolle4, 2024-07-17:
### Bambu Studio Version

Fresh Install

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04 LTS

### Additional system information

# System Details Report
---

## Report details
- **Date generated:**                              2024-07-16 20:00:03

## Hardware Information:
- **Hardware Model:**                              Gigabyte Technology Co., Ltd. Z370 AORUS Gaming 5
- **Memory:**                                      16.0 GiB
- **Processor:**                                   Intel® Core™ i5-8600K × 6
- **Graphics:**                                    Software Rendering
- **Disk Capacity:**                               2.0 TB

## Software Information:
- **Firmware Version:**                            F3
- **OS Name:**                                     Ubuntu 24.04 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.8.0-38-generic

### Printer

Bambu Lab P1S

### How to reproduce

I used 

`do-release-upgrade`

to upgrade Ubuntu 22.04 LTS to 24.04 LTS. I had configured BambuStudio on 22.04, but it was not working in 24.04. 

I recompiled it using the Linux compile guide ([https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)). Both the scripts and the full build resulted in similar errors (error shown is from the scripts method).

### Actual results

[BambuStudio-Linux24.04Install-Error.odt](https://github.com/user-attachments/files/16258007/BambuStudio-Linux24.04Install-Error.odt)

These are the last 4 pages of output from the build operation, claiming that wxWebViewer is insufficiently called through the cmake function. Two errors are thrown and the compile stops.

### Expected results

The install file should successfully compile, forming an appimage. 

### Project file & Debug log uploads

[BambuStudio-Linux24.04Install-Error.odt](https://github.com/user-attachments/files/16258881/BambuStudio-Linux24.04Install-Error.odt)

This is the last 4 pages of the output of the build.

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-07-18:
Have you updated the latest code? A number of commits have been added recently to resolve compilation issues on ubuntu24.04.
https://github.com/bambulab/BambuStudio/actions/runs/9967148202
This looks like a webview issue, make sure these commits already exist and rebuild the dependencies.
https://github.com/bambulab/BambuStudio/commit/a181250668fae3110e329c0dd289f1e71dd1bb45
https://github.com/bambulab/BambuStudio/commit/fe359394f3065867f3e84e806fec6bb1166357f0

CStolle4, 2024-07-19:
That latest code worked. Thanks so much!

closed 2024-07-19 by CStolle4