bambulab/BambuStudio #4511: Is there an existing issue for this problem?

ycarriger, 2024-07-17:
### Bambu Studio Version

1.9.3.5

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Series

### How to reproduce

1. Start a new project.
3. On a blank plate, right-click and select Add Primitive, then select Cube.
4. In the left-hand Process section, select the Objects tab.
5. Click on the plate name, for the "Other layers filament sequence" option, change it from Auto to Customize (no need to actually change the default order).
6. Click the Slice Plate button.
7. An Orca Slicer error pops up, "Serializing NaN".

### Actual results

Slicing fails when serializing the configuration. An error pops up, "Serializing NaN"
[image]

### Expected results

Slicer successfully completes slicing 

### Project file & Debug log uploads

[BBL Bug 1.zip](https://github.com/user-attachments/files/16258971/BBL.Bug.1.zip)
[debug_Tue_Jul_16_21_26_08_10080.log.zip](https://github.com/user-attachments/files/16258975/debug_Tue_Jul_16_21_26_08_10080.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

moe001, 2024-07-21:
我在相同版本上遇到了相同的问题，在切片一个包含有5种耗材颜色的小型模型中，但在我一些其他项目中没有出现。这个问题很令人恼火，因为我在制作一些具有特殊视觉效果的模型，需要在加工过程中反复手动更换耗材，如果不能自定义打印顺序，有时候没有足够的间隔让我替换ams中的料盘。对于一个商业公司来说，最近的更新修复速度太令人失望了。

moe001, 2024-07-22:
似乎在大部分情况下，手动设置左侧的两种层数为较小的数值可以解决这个问题，在bug没有修复之前，这可能是个比较好的折中方案

QingZhangBambu, 2024-07-26:
@moe001 您好，非常抱歉，现在在处于大版本的更新间隔期间，故而发布版本间隔较长。

orbitinstasis, 2024-08-01:
Same thing happens to me, when will this be fixed?

zhimin-zeng-bambulab, 2024-08-06:
[image]

This value is a built-in special value. You can change this value to "end" or other value ​to avoid this problem.

closed 2024-10-18 by zhimin-zeng-bambulab

lgarron, 2024-10-26:
> This value is a built-in special value. You can change this value to "end" or other value ​to avoid this problem.

Thank you, this post might just have saved me a lot of debuggin!