bambulab/BambuStudio #4512: Make LAN-only mode actually usable

UltraBlackLinux, 2024-07-17:
LAN-only mode is an absolutely crippled feature:
- [ ] #4026 
- [ ] #3210
- [ ] #3272
- [ ] #2232 
- [ ] #1491
- [ ] #1429
- [ ] #4295

Why would you prefer online features over local LAN features? I don't need (nor want) an app on my phone to control my printer, and I doubt I'm the only person with this opinion.

Most of the time when you're printing something you're probably within arm's reach of the printer. Why all these networking features? Why do I need to remotely move my printer's toolhead while on the road?
If I need to intervene with the printing process then I'll probably need to get physical access to the printer as well. You can't take a bad print off the build-plate using the toolhead, so what's the point?

(Speculation) Is this internet "requirement" just intended to allow Bambu to easily track almost everything the user does with their device?
Maybe I just don't want to share all of this...
[image]

Please, fix LAN-only mode so that it doesn't almost suspiciously often hold you back. I'm grateful that it does exist and won't turn the printer into e-waste if Bambulab ever seizes to exist but sadly it's currently barely usable.

timothyzalusky, 2024-07-19:
Come on Bambu Lab.  I don't think this is too much to ask.  I love your printers, but this makes it hard to recommend them to the company I work at.  Please fix ASAP.  This should be your one and only focus until complete.  Then add features, but not until this is done.  @walterwongbbl this is now approaching the two year mark of a lot of missing features/adding printers via IP address (available on all networkable competitor printers).

UBaer21, 2024-09-18:
There is still no official statement from BAMBU dev about this?  It seems these Issues need to be made more public.

SlothCroissant, 2024-09-22:
FYI, Adding printer by IP *should* land in 1.10 (ETA is likely November-ish from what I can tell?). Hopefully this puts us a big step forward in overall LAN-mode usability.

https://github.com/bambulab/BambuStudio/issues/2232#issuecomment-2310529584

vordenken, 2024-12-16:
The newest firmwares now allows upgrading to future version via the microSD card - this is huge! Thanks to the bambulab team for this.
Maybe you could also implement the feature to access the microSD card via bambuStudio (in lan-mode)? :)

UltraBlackLinux, 2024-12-16:
the first and only time in history I will upgrade my printer's firmware :)
Thanks for the info!

gamelaster, 2025-01-21:
Sadly, even the feature "add printer by IP" landed into BambuStudio, it's pretty useless, because unless you do not receive broadcast packet from the printer, printer does not show up in the list of devices (as reported in my issue #5833 ).

kwuite, 2025-05-13:
The KPN internet provider in the Netherlands cut off my internet in the evening. I came home from work excited to fix a print job that was failing for the last few days. But unfortunately today I faced the weakness of my bambulab x1 carbon printer. 

My printer is right next to the router.
I followed all online suggestions (researching via a 4G mobile phone network)

I tested my laptop over wifi and desktop over ethernet. Setup all network to private, disabled all firewalls and I can confirm Bambu Studio shows it actualy can connect, but the connection drops when trying to print.

The most I got out of 2 hours of trial and error was a 70% succesfull progress bar and then an error -4030.

I love this 3D printer but Lan Mode does not exist. This feature brings shame to such a beautifull device

With 18 years of professional IT background it should not be possible for me to not be able to print locally. I gave it my all and I know allot and this device is just failing over lan mode.

Did I unplug and reset and reinstall everything? Yes I did. I am out of options...

## Details.

Firmware bambulab: 01.09.00.01
AMS: 00.01.06.62
Bambu Studio: 02.00.03.04.54
OS: Windows 11 latest patch april 2025
Network: wifi 6 (laptop) wifi 4(bambu), 2.5Gb ethernet.