bambulab/BambuStudio #4515: Your nozzle diameter in sliced file is not consistent with the saved nozzle.

alessioCentamore, 2024-07-18:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

mac os 14.5

### Printer

a1 mini

### How to reproduce

I changed the extruder in the printer from a 0.4 I put a 0.6 and then I selected the extruder in bambu studio but when I start printing it gives me this message:
Please check the following:
1. Your nozzle diameter in sliced file is not consistent with the saved nozzle. If you changed your nozzle lately, please go to Device > Printer Parts to change settings.
nozzle in preset: 0.6
nozzle memorized: 0.4
Please fix the error above, otherwise printing cannot continue.

### Actual results

[image]

### Expected results

the print should start

### Project file & Debug log uploads

[cover.3mf.zip](https://github.com/user-attachments/files/16284595/cover.3mf.zip)
[log.zip](https://github.com/user-attachments/files/16284621/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

P3skyBee, 2024-07-27:
Same issue

QingZhangBambu, 2024-08-01:
@alessioCentamore it's a nozzle check to ensure the printer can print the model, you need to set the default nozzle diameter from 0.4 to 0.6.

toneman77, 2024-12-02:
> @alessioCentamore it's a nozzle check to ensure the printer can print the model, you need to set the default nozzle diameter from 0.4 to 0.6.

Where can you change the default nozzle size? 
I changed it in the printer settings "Extruder 1" from 0.4 to 0.6

domwrap, 2025-02-03:
> [@alessioCentamore](https://github.com/alessioCentamore) it's a nozzle check to ensure the printer can print the model, you need to set the default nozzle diameter from 0.4 to 0.6.

Sorry but you've just repeated what the error message says, but have not explained where it should be fixed. For someone that hasn't seen this message before it's not obvious enough that an issue has been raised (and I've found myself here trying to resolve it).

I added a 0.6 nozzle to my printer so can choose 0.4 or 0.6, I've 0.6 selected both for Printer and Process. Where should the "default" be changed/set to match the selection so the print can begin?

domwrap, 2025-02-03:
For anyone else that also finds themselves here:

[image]

Thanks to: https://www.reddit.com/r/BambuLab/comments/1aku0vb/comment/m53l94j/

closed 2025-02-11 by QingZhangBambu