bambulab/BambuStudio #4521: Brim Thickness / Multiple layers

adamfilip, 2024-07-18:
I would like the ability to add multiple layers of brim. to give objects a very secure footing. This is very useful when especially when printing a large quantity to prevent detachment

frank-pian, 2024-07-22:
@adamfilip Can you be more specific in describing your idea, or drawing a schematic?

jaredBevis, 2026-01-16:
I second a thicker brim option.  Not so much for stability/adhesion but rather to improve the plate cleanup.  Sometimes filament sticks too good on thin layers and It's a real pain to clean it up.  A single layer may not be enough material get it with a plastic scraping tool and I'm avoiding metal scraping as much as possible to save my build plates.

mirodecz-creator, 2026-10-26:
I support adding a configurable multi-layer brim, with two related improvements:

**1. Brim height: 1–5 layers**
Allow users to set brim height independently of brim width and brim-object gap. At a 0.2 mm layer height, this would provide a 0.2–1.0 mm thick brim. Additional layers should respect the object geometry and configured gap.

**2. Optional bridge/tab brim**
Allow narrow, breakaway bridges between the brim and the object, rather than requiring continuous contact around the perimeter. Please provide controls for bridge width, spacing/count, height, and manual placement.

An additional option could connect the brims of nearby objects with removable bridges.

The practical use case is printing tall, narrow parts, such as a small cylindrical shaft. A thicker brim could resist bending near the base, while controlled breakaway bridges could balance stability and easy removal.

These should be optional settings, preserving the current single-layer brim as the default.