bambulab/BambuStudio #4523: Auto-Refill does NOT work reliably.

Agoodpaddlin, 2024-07-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

NA

### Printer

X1C + AMS

### How to reproduce

Install 2 identical spools of filament into 2 slots of the AMS next to each other. A3 and A4.
Click on the AUTO-REFILL option in Bambu Studio and make sure its setup to go from A3 to A4 automatically when A3 empties.
Run print.

### Actual results

A3 reaches the end of the filament roll and pauses asking for a new roll of filament to be inserted into A3.

### Expected results

The printer should automatically recognise A3 has runout and load A4 to continue printing.

ALSO, WHEN YOU CLICK TO REPORT A PROBLEM AND BAMBU STUDIO SAYS IT NEEDS TO VERIFY YOU, PERHAPS TELL THE USER THAT THE REQUEST IS COMING FROM GITHUB AND NOT BAMBU. THIS IS BASIC STUFF GUYS. PAY ATTENTION AND TURN YOUR BRAINS ON.

### Project file & Debug log uploads

[debug_Tue_Jul_16_20_10_03_15772.log.zip](https://github.com/user-attachments/files/16302217/debug_Tue_Jul_16_20_10_03_15772.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-08-08:
@Agoodpaddlin 
We have provided feedback on this issue

adx-git, 2024-12-06:
Just got the same issue, is there a solution?