bambulab/BambuStudio #4525: 3MF File Contains NO GEOMETRY

JMarsden92, 2024-07-19:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

HP All In One
Intel(R) Core(TM) i7-10700T CPU @ 2.00GHz   2.00 GHz
16gb ram

### Printer

Bambu P1S

### How to reproduce

[image]

3MF File is huge... Can't attach...

[debug_Fri_Jul_19_03_10_22_15968.log.zip](https://github.com/user-attachments/files/16309511/debug_Fri_Jul_19_03_10_22_15968.log.zip)

### Actual results

I was working on a file in Bambu Studio.. Imported a new piece of the file and saved...  when i went to reopen it gives the "Geometry" issue...   The file is LARGE...
I am unsure if there is anything I can do to remove the LAST added piece...
I changed it to a ZIP, but unsure how to tell which piece is the last piece...

### Expected results

File should have opened but give Geometry issue...
[image]

### Project file & Debug log uploads

[image]
[debug_Fri_Jul_19_03_10_22_15968.log.zip](https://github.com/user-attachments/files/16309528/debug_Fri_Jul_19_03_10_22_15968.log.zip)
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-07-22:
hello，Can you provide this 3mf file？thanks

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00