bambulab/BambuStudio #4527: Quit hotkey doesnt work on linux (ctrl+q)

zlice, 2024-07-19:
### Bambu Studio Version

Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.AppImage

### Where is the application from?

Bambu Lab github releases

### OS version

void linux 6.10

### Printer

X1C

### How to reproduce

- start bambu studio
- ctrl+q to quit (nothing)
- import file
- ctrl+q to quit (nothing)

### Actual results

doesn't quit

### Expected results

hotkey quits normally

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-07-24:
It seems that this hotkey is not added manually; it might be the default behavior of wxWidgets.
https://github.com/bambulab/BambuStudio/blob/90417112c1aefb0c951cee6e9fd5326cbd0d8e0a/src/slic3r/GUI/MainFrame.cpp#L2413

MackBambu, 2024-07-24:
Same issues
https://github.com/prusa3d/PrusaSlicer/issues/13013

zlice, 2024-07-24:
Thanks, will close as dupe.