bambulab/BambuStudio #4528: 空白

wenkeWQQ, 2024-07-20:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

X1-Carbon

### How to reproduce

[image]

### Actual results

安装完成向导配置空白

### Expected results

正常配置

### Project file & Debug log uploads

[Uploading log.zip…]()

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-07-23:
重新安装 WebView2 运行时试试。
安装程序在 Studio 安装目录下的 plugin 子目录。

wenkeWQQ, 2024-07-23:
已经安装过了还是一样的

wenkeWQQ, 2024-07-23:
用测试的程序试过了还是会报错
[image]