bambulab/BambuStudio #4534: Overly Thin Scroll Bars!

sonjamichelle, 2024-07-21:
**Is your feature request related to a problem? Please describe.**
Why is it that you feel the need to follow the obscene trned of continullay thinning scrollbars????
Not all of us have touch screens, use our phones/iphones or tablets for all of our computing needs. Some of STILL ACTUALLY use REALL computers!!
Even with a touch screen my tremor and neuropathy makes touch screen chellenging and frustrating, same goes for laptop touchpads.

**Describe the solution you'd like**
The abilit to either choose thick or thing scrollbars or the ability to set default/thin/thick or custom width.

**Describe alternatives you've considered**
Read the above!!!

**Additional context**
STOP FOLLOWING ALL THE TRENDS!!! USE YOUR OWN BRAINS AND IDEAS!!!!!

sonjamichelle, 2024-07-21:
The scrollbar flippin thin that I could NOT scroll down any further to select addtional filaments.

[image]

Each time I pressed the right button on the touchpad to "drag" down the "scrollbar" it select the nearest filament. I had to go hunt down a wired mouse with a scroll wheel!

pgifford, 2024-08-28:
The problem is not only that it's thin but also that it is only an indicator, i.e. the scrollbar is not draggable and should be.  I have a lot (a LOT) of filament presets and it is a chore to scroll down with the wheel when I want to select a filament from Ziro.  Typing a letter does not jump to a selection that starts with the letter typed.  Arrow keys and Page Up/Page Down keys don't work.

Scrollbars in Bambu Studio should enable scrolling and not merely serve as an indicator.  The select box should respond to keyboard input for convenience and accessibility.

sonjamichelle, 2024-10-03:
With a trackpad, the filament list is useless! You can't tap, hold and scroll. It just selects what is under the cursor!

Many times, when working remotely it is not convenient to have an attached mouse. eg: No table or desk space, working from a vehicle, on your lap, hence the name LAPTOP.