bambulab/BambuStudio #4535: Opening 3D model in bambu studio from makerworld.

EmilSokolowski, 2024-07-22:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro 22H2 build 19045.4651

### Additional system information

AMD Ryzen 7 3800XT 8-Core Processor               3.89 GHz
16,0 GB
Windows 10, 64-bits 

### Printer

babulab P1S with AMS

### How to reproduce

Open Bambu studio
Go to Makerlab. (opens in Firefox on my PC)
[image]
Choose a 3D model (any model will do) , select printer P1S and  klick "open in Bambu studio"
[image]

### Actual results

Then Bambu studio comes to the front in "Home" screen but nothing opens.
[image]
No Project is opened.

### Expected results

That a project is opened in Bambu studio, without closing Bambu studio first.
Because when you close bambu studio before you click on "open in babu studio" then it works perfect.

### Project file & Debug log uploads

[Bambustudio-log.zip](https://github.com/user-attachments/files/16328118/Bambustudio-log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Shreerang912, 2026-09-30:
Even After 2 years, This EXACT Bug is happening to me

EmilSokolowski, 2026-10-03:
> Even After 2 years, This EXACT Bug is happening to me

Yep, they still havent changed a thing about this.

Shreerang912, 2026-10-03:
Has someone found out why exactly this handshake between maker world and bambu studio fails. I wonder if it is possible to fix this by making some browser extension if he problem is in the browser side. 
Tbh I would like to have almost all features of maker world especially comments and collection on bambu studio.

Shreerang912, 2026-10-03:
I found it.  THE FIX.  Ctrl + P on Bambu studio to open preferences then turn OFF the "**Keep only one Bambu Studio instance**".
Restart Bambu studio. 
Voila! Just one simple setting, lol.