bambulab/BambuStudio #4536: Slicer craches

Hares42, 2024-07-22:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU : i9 13900KF
RAM : 32 GO 6200Mhz
GTU :  RTX 4900

### Printer

Bambu Lab X1C

### How to reproduce

It craches when i try to slice (the software just shutdown without any message).
It seems to happen with big 3d models

### Actual results

Nothing

### Expected results

It should slice the model

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16337219/log.zip)
[majin-vegeta-bust-2.zip](https://github.com/user-attachments/files/16337256/majin-vegeta-bust-2.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SamppaD78, 2024-07-27:
I have the same issue,slicing big models,program start to slice the model and at 15% comoleted it crashes,(software just closes without any error masage)  Windows 10 64 bit,all latest updates installed ,intel i5-10300H,32gb of memory. Nvidia  GTX1650 sstudio driver 560.70 .

QingZhangBambu, 2024-08-01:
@Hares42 @SamppaD78 
https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/
It's an intel software bug that's been going on for a while now.
Turning off the overclocking settings on the motherboard will improve it.

https://github.com/bambulab/BambuStudio/issues/3318
https://github.com/prusa3d/PrusaSlicer/issues/11845

Hares42, 2024-08-01:
Hello,

 

I never overclocked my computer.

So i don’t know what i could disable on my BIOS to improve the stability of the slicer.

 

Cordialement,

 

Jean-Jacques HERMITTE

 

https://www.xda-developers.com/intel-cracks-the-cause-raptor-lake-instability/

#3318 <https://github.com/bambulab/BambuStudio/issues/3318> 
prusa3d/PrusaSlicer#11845 <https://github.com/prusa3d/PrusaSlicer/issues/11845>

QingZhangBambu, 2024-08-14:
@Hares42 then, may you could try this methods
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

Hares42, 2024-08-14:
Hello,

 

It seems to work with one of the model which didn’t worked.

I’ll try with others.

 

Thanks for the help.

 

Cordialement,

 

Jean-Jacques HERMITTE

 

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing