bambulab/BambuStudio #4538: Text in dialogs is shown as empty squares

joepadmiraal, 2024-07-23:
### Bambu Studio Version

1.9.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

X11
Gnome 46
Mutter 46.2-1ubuntu0.24.04.1
Kernel Linux 6.8.0-38-generic

### Printer

P1S

### How to reproduce

1. open bambu studio
2. load a model
3. click on the model

or click on slice plate

### Actual results

[image]

[image]

### Expected results

I would expect letters that I can read instead of squares ;)

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/16353294/log.zip)
[log.zip](https://github.com/user-attachments/files/16353329/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-07-26:
hi @joepadmiraal,
This looks like imgui font atlas texture fails to upload to GPU.
I need some info to reproduce this：
1. GPU information
2. using virtual machine？Enable graphics acceleration？

frank-pian, 2024-07-26:
> I now noticed that when I plug in a monitor, the texts suddenly appears. Even without restarting Bambu studio, and with Bambu studio still showing on my laptop display. Unplugging the monitor makes the text disappear again. I'm not using a virtual machine.

Some laptops may switch to discrete graphics output when you plug in the monitor. Is the text displaying fine when you plug in the monitor, using an NVIDIA GPU?

joepadmiraal, 2024-07-26:
I now found that when I set scaling to 200% instead of 225% the text also looks good.
So thats without the second monitor.

frank-pian, 2024-07-26:
> I now found that when I set scaling to 200% instead of 225% the text also looks good. Without using the second monitor. `nvidia-smi` does not show the bambu studio process, before and after changing the scaling, so its running on the Intel GPU and the issue seems purely scaling related.

Thanks, I'll get a 4K monitor and try to reproduce the issue.

MackBambu, 2024-08-04:
[image]
[image]
I did some testing in a VM and I suspect it may still have something to do with the gpu

joepadmiraal, 2024-08-05:
Interesting, this is with 24.04?
I can try to reproduce it in a VM for you in about 2 weeks when I have access to a normal computer again.

joepadmiraal, 2024-08-22:
Did you also test under X11?

I could reproduce it on a different machine:
Intel NUC7i7DNHE
Only 1 monitor attached
GPU: Intel UHD Graphics 620
Resolution: 3840x2160
Using X11
Fractional scaling: 225% 
Fresh installation of Ubuntu24.04 with all apt updates applied

I also tried to create a VM and gave up after more than 10 attempts.
- on my laptop via Gnome Boxes, I can run Ubuntu but I'm unable to configure the resolution and scaling I need.
- on 2 different NUC's with different Ubuntu versions as host via VirtualBox, Ubuntu 24.04 installer just hangs
- on a M2 Macbook via VirtualBox, found out that Bambu Studio does not run on ARM
I tried a lot of variations of Virtualbox graphical settings, but none seem to get me past the installer.

MackBambu, 2024-09-08:
Thanks, can be reproduced on x11.
You could try working under wayland first, it seems to be fine.